eutelescope/eutelescope #472: CLHEP tarball changed address

mbomben, 2019-06-24:
Dear all,
I have tried today to install eutelescope but it fails; the reason is that the CLHEP tarball location as changed.
It was:

> http://proj-clhep.web.cern.ch/proj-clhep/DISTRIBUTION/distributions/clhep-2.3.1.1.tgz

It has been changed to:

> http://proj-clhep.web.cern.ch/proj-clhep/dist1/clhep-2.3.1.1.tgz

Could you please fix this?

Many thanks in advance and king regards,
Marco Bomben

edorossi, 2019-06-28:
Hi Marco,

I fixed the link in the ILCInstall repository and I did some quick tests and it looks like it is working fine. Let me know if this fixes your installation! 

Cheers,

Edo

mbomben, 2019-06-30:
Hello Edo,
thanks. I have tried it and it works again.
Thanks a lot,
Marco Bomben

closed 2019-06-30 by mbomben