eutelescope/eutelescope #473: Reintroduce EUTelProcessorGeometricClustering

tbisanz, 2019-07-01:
We should rename it to `EUTelGeometricClustering` and update the AconiteQuadFEI4_DAF example for the ATLAS pixel data to use it.

edorossi, 2019-07-01:
Hi Tobias,

The only issue I have is that the geometric clustering processor is not so well documented, so it can be confusing to use. From the name, it looks like it is just a different clustering algorithm, but in reality is effectively implementing the geometry already, unlike the sparse clustering. In case of rotations in the gear, the difference between geometric and sparse clustering can be quite large, which one would not expect from just the processor names.
Do you have any idea on how to make it more understandable for new users?

Cheers,

Edo

jha92, 2019-07-02:
Hi Tobias, hi Edo,
in principle, I am also in favor to reintroduce it, but as Edo pointed out, it has to be clear. Especially, as we are currently figuring out a clear way also for GBL to handle the different kinds of geometry implementations (e.g. rotations). So maybe, when we are clear with this, it could be a good point to reintroduce?
Cheers,
JH

tbisanz, 2019-07-02:
The clustering processor should be independent of any rotation scheme - so I don't think it has to be attached to that issue. Regarding that however, I would make sure not to introduce some conflicting schemes or conventions as this would fuck up quite a lot!