eutelescope/eutelescope #478: No EUTelMille processor after new installation

maweer, 2019-07-15:
Hello all,
I'm currently working on a new example for ATLAS pixel people to reconstruct data taken with Rd53a. The example is based on the aconite4chip_local example and uses daf. 
In  the alignment step I get the error message "EUTelMille" is a unknown processor. But when I look into "eutelescope/processors/src/legacy/ " there is a file called EUTelMille.cc.
This seems to be a new, because with the old millepede: external/millepede2/V04-03-09 the processor was there. 
But with the new installation of Eutelescope (from 4.7.19): external/millepede2/V04-05-01 the processor is not there any more.
Why is the processor removed?

jha92, 2019-07-16:
Hi,
should be fixed now. Please have a try. Same for @mkofarag, EUTelMille should now work again.
Cheers,
Jan-Hendrik

closed 2019-07-16 by jha92

jha92, 2019-07-16:
Had to add another bugfix, now it should compile.

mkofarag, 2019-07-19:
Hi Jan-Hendrik,

Thanks, it builds fine now.

Cheers,
Monika