eutelescope/eutelescope #481: Wrong Z in the GBL dumped hits (problems for rotated DUT)

terzo, 2019-08-13:
In the file "preprocess/src/EUTelGBL.cc" the impact point of the track on the DUT plane is extracted at a fixed nominal Z position. 

This does not take into account of eventual rotations of the DUT planes and ends up giving significant errors of the impact point when moving to the local frame.

For rotated planes the output of GBL is thus wrong and results in problem with the analysis.

In the code there is a comment that suggests to fix this by using the Z position of the hit on the DUT (if this exist). I have instead an alternative more precise calculation that holds even in case no hit on the DUT is available. 
In the following the intersection between the uptriplet and the DUT is calculated using Eigen functions. The old code is commented at the end for comparison.

```
 if(_dumpTracks) { //CHECK ME CAREFULLY
             auto hit_a = uptriplet.getpoint_at(0);
             auto hit_b = uptriplet.getpoint_at(_planePosition[ix]);
             Eigen::Vector3d a(hit_a.x,hit_a.y,hit_a.z);
             Eigen::Vector3d b(hit_b.x,hit_b.y,hit_b.z);
             Eigen::Vector3d normal =   geo::gGeometry().getPlaneNormalVector(_sensorIDVec[ix]);
             Eigen::Hyperplane<double,3> pl(normal,Eigen::Vector3d(0.0,0.0,_planePosition[ix]));
             Eigen::ParametrizedLine<double,3> line = Eigen::ParametrizedLine<double,3>::Through(a,b);
             Eigen::Vector3d point = line.intersectionPoint(pl);
             double zz = point[2];
          
            thisTrack->setIntVal(0, _sensorIDVec[ix]); //sensor ID is an int
             thisTrack->setIntVal(1, Ndf); //Ndf is an int
             thisTrack->setIntVal(2, numbertracks);
             thisTrack->setFloatVal(0, Chi2); //chi2
             thisTrack->setFloatVal(1, uptriplet.getx_at(zz) + localPar[3]); // x track position (global system)
             thisTrack->setFloatVal(2, uptriplet.gety_at(zz) + localPar[4]); // y track position (global system)
             thisTrack->setFloatVal(3, zz); // z of the impact point on the plane
             //thisTrack->setFloatVal(1, uptriplet.getx_at(_planePosition[ix]) + localPar[3]); // x track position (global system)
             //thisTrack->setFloatVal(2, uptriplet.gety_at(_planePosition[ix]) + localPar[4]); // y track position (global system)
             //thisTrack->setFloatVal(3, _planePosition[ix]); // z of the plane, FIXME: we should use z hit position if `posible`
```
Please consider of implementing this fix also in the tag versions and improve it if necessary.

Here some examples obtained with the TBMon2 analysis tool
Before Fix:
[image]

After Fix:
[image]

Best,
Stefano

jha92, 2019-09-04:
Hi Stefano,
first of all, sorry that we did not replied (summer break, own testbeam campaigns etc).
I already had a look at your PR and for me the solution looks very fine. So I will merge it soon.
Why do you want to have it in the tagged versions, or in which specifically?
I have a question to your provided TBMon2 plots. I can see the effect before he fix of the rotation in the Y residuals, which is gone after fixing. I am only wondering about the scale of the plots after fix, where you see huge outliers (over 1000um). Do you have an explanation for these?
Cheers,
J-H

terzo, 2019-09-06:
I would like to have it in a tagged version because for the ITk pixel
Market Survey we will need to have a reference version to use which will
not change in order to compare the results of the different vendors.
Version 2.0 or a new one is both fine.

It may just be noise or other hits not belonging to the actual tracks.
These plots are made with raw data as a check before the actual analysis
with a matching radious large enough to allow the inclusion of possible
large misalignment.

Cheers,
Stefano

jha92, 2019-09-18:
Hi Stefano,
I approved now the PR, sorry for the delay.
We still have some open projects, which we would like to merge as well. What is the time scale you would like to have a tagged version? In general, we were planning to provide now more often tagged versions.
Cheers,
J-H

terzo, 2019-09-18:
Hi,

thanks!
For the tagged version the sooner the better. But in the meanwhile we can
use the master of course.

Cheers,
Stefano
--
Stefano Terzo
MPP ATLAS Inner Detector Group
Max-Planck-Institut für Physik (Werner-Heisenberg-Institut)
Room 207
Föhringer Ring 6
D-80805 München, Germany
Phone: +49 89 32354-416
eMail: stefano.terzo@mpp.mpg.de

closed 2019-11-18 by jha92