eutelescope/eutelescope #483: Gear File update for multiple DUTs

AlexDroll, 2019-09-04:
Hi all, 
I try to make it short. When using the EUTelPreAligner or EUTelGBL processor I have problems in aligning multiple DUTs (in my case 4). 
I checked the different steps and by the console output the alignment seems to work fine. What happens is that when the new Gear files are written, the DUT layer get mixed up. In my case the DUTs have the IDs 30-33. And in the new gear file the coordinates of the layers 30 and 32 (31 and 33) are exchanged when the GEAR file is written (I marked the position where this happens in the code). Since gear is not part of this repo, it's probably a bug/unknow feature from the GEAR side. 
Did you have similar problems before? 
Thanks for your help! 
Alex

https://github.com/eutelescope/eutelescope/blob/2ceafa38901bea9e4d27cd370cc79dab4bd4abc5/processors/src/EUTelPreAligner.cc#L431

AlexDroll, 2019-09-04:
I found my fault in the first gear file. Sorry everything working fine!

closed 2019-09-04 by AlexDroll

jha92, 2019-09-04:
Hi Alex,
good to hear that the problem solved itself :)
In general, I heard once that there can be indeed a mix up in the case that DUTs are defined at the same z-position (for technical reasons done for ATLAS ITk Strips), but there the solution would be then only a slight difference in z. But apparently your problem was different.
Cheers,
J-H

AlexDroll, 2019-09-04:
No, I have also a design where the DUTs are in pairs at the same z position. I also changed the z position slightly to have a defined z order. The problem was that the z shift was not large enough.