eutelescope/eutelescope #484: Unable to run the example code - just at the first step

jnsandhya, 2019-11-18:
Dear experts, 

I am new to EUTelescope. The installation from the EUTeslescope page here : 
http://eutelescope.web.cern.ch/content/installation 
worked out of the box. 
Now following this: 
http://eutelescope.web.cern.ch/content/job-submission
I am trying to run the example using the data from oct-2016 test beam for the CMS Tracker detector. I understand that corresponding to name of the global sections from config, there should be an xml file in the steering path. However comparing to most recent files, I noticed that some processor names changed and after updating the names - I got rid of the errors coming from unknown names but now at the step : 
$EUTELESCOPE/jobsub/jobsub.py --config=${ANALYSIS_CONF}/config.cfg -csv ${ANALYSIS_CONF}/runlist.csv converter 103 -g

It reports this error in the end : 
10:26:49 jobsub.converter(INFO): [ MESSAGE5 "Save"] Writing the output file /home/sjain/ILCSoft/v01-19-02/Eutelescope/master/jobsub/examples/cms-1FEI4-noDUT_Oct16/output/lcio/run000103-converter.slcio
10:26:49 jobsub.converter(INFO): [ MESSAGE5 "UniversalNativeReader"] Successfully finished
10:26:50 jobsub(ERROR): Marlin returned with error code -11
10:26:50 jobsub(WARNING): There were 1 error messages reported

And I am not sure, where from this could be coming. 
Any help here would really be appreciated. Please let me know if any further information is needed from my side. 

Thanks a lot and Best regards
Sandhya

jha92, 2019-11-18:
Dear Sandhya,

we are happy to help you to get started with EUTelescope :)
First of all: sorry, but the website is by now pretty much outdated and maybe we are planning to remove it, because the most up-to-date instructions etc should be here at the github page.

Therefore also, I am not sure which version you have installed, as also a lot of stuff has changed in the ilcinstall. For this, please have a detailed look [here](https://github.com/eutelescope/iLCInstall) and checkout the README for the instructions.

Next are then the examples, where currently only a subset is maintained, as some of the old ones are actually also pretty old. Do you really need the CMS-Tracker-Example? The normal startup is to run the examples GBL_noDUT, GBL_SUT or GBL_DUT. If you have special needs, please tell us.

I hope it is okay to not go for now in a detailed debugging of your above mentioned error, but first start on the same ground, if there is no problem for you to do so.

Cheers,
Jan-Hendrik

jnsandhya, 2019-11-18:
Dear Jan-Hendrik, 

Thanks a lot for replying back quick. I can definitely try to install it from the beginning if needed.
But just to be complete on what I used to install are these following commands : 
> git clone -b dev-base https://github.com/eutelescope/ilcinstall $ILCSOFT/ilcinstall
> cd $ILCSOFT/ilcinstall
> ./ilcsoft-install -i examples/eutelescope/release-standalone.cfg

Which might be different from the master branch mentioned in readme: 
> git clone -b master https://github.com/eutelescope/ilcinstall

Please let me know, if it is needed to switch to master branch since you might know the changes well and install again. 
 
The idea to use CMS TRACKER example is to be able to reproduce the tracker reconstruction and gain experience in that for the coming test-beam. 
I could have run GBL_DUT - this is where from I know the processor names changed for example but in absence of data file for this setup I am not sure how I should proceed. 
   
Thanks again for the help and looking forward to more. :) 

Best regards
Sandhya

jha92, 2019-11-18:
Hi Sandhya,

I guess for the standalone version, your setup is perfectly fine. But in general, the master as default is the preferred version. So maybe you can try to install the master version in parallel to your current version (as it takes some time...).
For example, one issue could be already that "error code -11" which can be in fact no real error (I think the output should have been created...).

So I infer you work for CMS? Was the CMS example in an older EUTel version publicly accessible (because I cannot find out anything about it)?
You mean with data file you do not have the raw files available? So then I guess you do not have AFS access, but for this we provide the files for the examples GBL_* [here](https://desycloud.desy.de/index.php/s/RxrpCyQrMJycct5).

Regards,
Jan-Hendrik

jnsandhya, 2019-11-18:
Hi Jan_Hendrik, 

Yes I am on it - to install the master branch to make sure we are on the same ground atleast. 
Secondly, yes I work for the CMS experiment. And the data is not really publically available but I also think it is fine to share one of the raw files to be able to follow up what is ongoing. The raw file can be obtained from here : 
https://cernbox.cern.ch/index.php/s/q4M38owSs0njpKY

Yes, I was wondering about the raw files only. Thanks a lot for pointing me to them. I can try them with fresh install and get back. 

Thanks again and best regards
Sandhya

jnsandhya, 2019-11-18:
Hi again, 

So, the fresh installation finished well but when I tried to run the example from GBL_DUT with the example file you pointed before, I get the following error: 
> $EUTELESCOPE/jobsub/jobsub.py --config=${ANALYSIS_CONF}/config.cfg csv ${ANALYSIS_CONF}/runlist.csv converter 2365
13:36:30 jobsub(ERROR): The list of runs contains non-integer and non-range values: '/home/sjain/cernbox/ILCSoft/eutel_v02-00-00/Eutelescope/master/jobsub/examples/GBL_DUT/runlist.csv'

And I get the same error if I now try to run the CMS file as well. What is expected in the runlist as such? Also, since you mentioned the EUTelescope page is old, can we use the instructions mentioned on the page for running the examples? 

Thanks and Best regards
Sandhya

jha92, 2019-11-18:
Hi,

yes, please follow the accompanying README file. So, for example in this case, there is no converter step at all, as the data was converted already due to EUDAQ2 data taking. So it starts with the plain noisypixel-step.

I guess the error comes due to a missing "-" in front of `csv`, so please simply use the following (as mentioned in README):
`jobsub -c config.cfg -csv runlist.csv -g noisypixel 2365`

jnsandhya, 2019-11-18:
Ah perfect, even the way one passes the command changed. Sorry was looking at a wrong README. Great, it works now. Thanks so much. :)

jha92, 2019-11-18:
No, the `-csv` was there always I think :)