eutelescope/eutelescope #490: Can‘t connect to https://svnsrv.desy.de/

chenzhl2018, 2022-07-18:
Hello，
When I down GEAR, there is a question: it could not connect to server (https://svnsrv.desy.de).How should I do?
Here is the log:
*** Downloading sources for GEAR version v01-06-eutel-pre...
********************************************************************************
svn download cmd: svn checkout https://svnsrv.desy.de/public/gear/gear/branches/v01-06-eutel-pre v01-06-eutel-pre
svn: E175002: Unable to connect to a repository at URL 'https://svnsrv.desy.de/public/gear/gear/branches/v01-06-eutel-pre'
svn: E175002: OPTIONS of 'https://svnsrv.desy.de/public/gear/gear/branches/v01-06-eutel-pre': could not connect to server (https://svnsrv.desy.de)

*** ERROR in module [ GEAR ]: DEBUG INFO: {'PLATFORM': 'Linux - CentOS Linux release 7.8.2003 (Core)', 'GCC_VERSION': 'gcc (GCC) 4.8.5 20150623 (Red Hat 4.8.5-39)', 'LSB_RELEASE': 'LSB Version:\t:core-4.1-amd64:core-4.1-noarch:cxx-4.1-amd64:cxx-4.1-noarch:desktop-4.1-amd64:desktop-4.1-noarch:languages-4.1-amd64:languages-4.1-noarch:printing-4.1-amd64:printing-4.1-noarch\nDistributor ID:\tCentOS\nDescription:\tCentOS Linux release 7.8.2003 (Core)\nRelease:\t7.8.2003\nCodename:\tCore', 'UNAME': 'Linux ui01.hep.ustc.edu.cn 3.10.0-1127.8.2.el7.x86_64 #1 SMP Tue May 12 16:57:42 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux'}

*** ERROR in module [ GEAR ]: Problems ocurred downloading sources with svn!!

Logfile for failed module: /ustcfs2/zchen/eutel_v02-00-00/.ilcinstall-logs/release-local-standalone-2022-7-18_3-54.log

Thanks in advance.

jha92, 2022-07-18:
Hi Chen,
may I ask you for which project you want to use EUTel and if it is already in use or if it is at the start?
The reason for this is that we are not supporting/maintaining EUTel widely anymore and urge users rather to switch to [Corryvreckan](https://gitlab.cern.ch/corryvreckan/corryvreckan/).
In case you really need EUTel, I can have another look on your issue, but you might consider the mentioned points above.

chenzhl2018, 2022-07-27:
Hi，
When I install EUTel，there is an error followed:
What should I do to deal with it？
Thanks in advance.

/scratch/ilcsoft/FastJet/3.4.0/build
Making install in src
make[1]: Entering directory `/scratch/ilcsoft/FastJet/3.4.0/build/src'
/bin/sh ../libtool  --tag=CXX   --mode=compile g++ -DHAVE_CONFIG_H -I. -I../../FastJet/src -I../include/fastjet    -g  -Woverloaded-virtual  -DDROP_CGAL  -I../../FastJet/src/../include -std=c++17 -MT libfastjet_la-DnnPlane.lo -MD -MP -MF .deps/libfastjet_la-DnnPlane.Tpo -c -o libfastjet_la-DnnPlane.lo `test -f 'DnnPlane.cc' || echo '../../FastJet/src/'`DnnPlane.cc
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I../../FastJet/src -I../include/fastjet -g -Woverloaded-virtual -DDROP_CGAL -I../../FastJet/src/../include -std=c++17 -MT libfastjet_la-DnnPlane.lo -MD -MP -MF .deps/libfastjet_la-DnnPlane.Tpo -c ../../FastJet/src/DnnPlane.cc 
g++: error: unrecognized command line option ‘-std=c++17’
make[1]: *** [libfastjet_la-DnnPlane.lo] Error 1
make[1]: Leaving directory `/scratch/ilcsoft/FastJet/3.4.0/build/src'
make: *** [install-recursive] Error 1
Traceback (most recent call last):
  File "./ilcsoft-install", line 128, in <module>
    ilcsoft.makeinstall()
  File "/ustcfs2/zchen/test/iLCInstall/ilcsoft/ilcsoft.py", line 505, in makeinstall
    mod.install([])
  File "/ustcfs2/zchen/test/iLCInstall/ilcsoft/baseilc.py", line 889, in install
    self.compile()
  File "/ustcfs2/zchen/test/iLCInstall/ilcsoft/fastjet.py", line 90, in compile
    os.chdir( self.installPath+"/fjcontrib-"+self.fjcontrib_version )
OSError: [Errno 2] No such file or directory: '/scratch/ilcsoft/FastJet/3.4.0/fjcontrib-1.046'